openhwfoundation/core-v-verif #2542: cv32e40x

vinomutty, 2024-10-16:
HI
we are trying to run cv32e40x 
 corev_rand_arithmetic_base_test , but we are facing the error.

[image]

vinomutty, 2024-10-16:
command:
make gen_corev-dv test TEST=corev_rand_arithmetic_base _test SIMULATOR=vsim USE _ISS=NO

hello word also not working

[image]

silabs-robin, 2024-10-16:
>SIMULATOR=vsim 

If you are using Questa, why does the screenshot reference Verilator?

--------

> hello word also not working

If I was having this problem, I would debug hello-world rather than corev_rand_arithmetic_base, as I think that would be easier (not by much, but a little).

--------

> mkdir: missing operand

This too indicates some env var is not set. Did you investigate my previous remark about this?

--------

Another note, IIRC `cv32e40x/release` might be more up-to-date when it comes to `vsim`-compatibility.
(But, the env var thing seems much more likely to be a culprit.)

MikeOpenHWGroup, 2024-10-16:
The last time I attempted to use Verilator with the E40X it would not compile.  The E40X RTL incorporates SystemVerilog Interfaces and at the time Verilator did not support them.  I believe the latest stable version (v5.028) does, but I have not attempted it.  I can guarantee that Verilator cannot support the UVM environment implemented in CORE-V-VERIF.

vinomutty, 2024-10-17:
we cloned new branch cv32e40x/release and tried running hello word testcase and getting this error
[image]

[image]

MikeOpenHWGroup, 2024-10-17:
There are two problems here:
1. You are using Verilator.  AS both @silabs-robin and I have indicated, it is very unlikely that Verilator can compile the E40X.
2. Because of the the above problem, the testbench in `core-v-verif/cv32e40x/tb/core` has not been maintained.  You will need to update it to match the pinout of the CV32E40X.

If you are willing to take this task on will can consider accepting a pull-request from you to get Verilator working with the E40X.  However, there is no guarantee of success since we do not know if Verilator can support all the RTL code-constructs used.  Do you have access to a commerical SV simulator?

vinomutty, 2024-10-26:
we cloned new branch with
git clone https://github.com/openhwgroup/core-v-verif/blob/cv32e40s/dev-1

We tried to run Hello-world test but facing the below errors

-- Compiling module riscv_random_stall
** Error: /Projects/marmik_project/hemashri.bhagavati/cv32e40s/core-v-verif/cv32e40s/tb/core/tb_riscv/riscv_random_stall.sv(140): (vlog-2730) Undefined variable: 'STANDARD'.
** Error: /Projects/marmik_project/hemashri.bhagavati/cv32e40s/core-v-verif/cv32e40s/tb/core/tb_riscv/riscv_random_stall.sv(142): (vlog-2730) Undefined variable: 'RANDOM'.
** Error: /Projects/marmik_project/hemashri.bhagavati/cv32e40s/core-v-verif/cv32e40s/tb/core/tb_riscv/riscv_random_stall.sv(191): (vlog-2730) Undefined variable: 'STANDARD'.
** Error: /Projects/marmik_project/hemashri.bhagavati/cv32e40s/core-v-verif/cv32e40s/tb/core/tb_riscv/riscv_random_stall.sv(193): (vlog-2730) Undefined variable: 'RANDOM'.
** Error: /Projects/marmik_project/hemashri.bhagavati/cv32e40s/core-v-verif/cv32e40s/tb/core/tb_riscv/riscv_random_interrupt_generator.sv(27): (vlog-13006) Could not find the package (perturbation_defines).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.

MikeOpenHWGroup, 2024-10-26:
It is doubtful that the `vsim.mk` file is fullly up-to-date.  Also, please ensure that you are using a fully featured version of Questa.