openhwfoundation/core-v-verif #2544: [cv_dv_utils] File list provided is not correcte

khandelwaltanuj, 2024-10-21:
Hello,

the file list provided to compile the uvm agents of cv_dv_utils is not correct. 

Regards
Tanuj

MikeOpenHWGroup, 2024-10-22:
Hi @khandelwaltanuj, do you mean [lib/cv_dv_utils/uvm/Files.f](https://github.com/openhwgroup/core-v-verif/blame/4e17969cfd7517dd0e26d4479da28a17220d8a4d/lib/cv_dv_utils/uvm/Files.f#L2)?  What error message do you get?

khandelwaltanuj, 2024-10-23:
Hi @MikeOpenHWGroup 

Yes the error is from this file. 

This file has following package which we have not provided with cv_dv_utils. They were for a specific project I believe. 

${CV_DV_UTILS_DIR}/uvm/dram_monitor/dram_monitor_pkg.sv
${CV_DV_UTILS_DIR}/uvm/dram_monitor/user_vif.sv
${CV_DV_UTILS_DIR}/uvm/dram_monitor/simple_dram_vif.sv

Thanks and Regards
Tanuj Khandelwal

MikeOpenHWGroup, 2024-10-23:
Yes, I believe you are right @khandelwaltanuj, thanks for pointing this out.

Hi @ludovicpion, can you confirm that these entries can be removed?

ludovicpion, 2024-10-24:
Hi @MikeOpenHWGroup,

Yes I confirm. We (CEA) concidered the dram_monitor has not mature enough (and we did not want to support it) to include it within cv_dv_utils.

Regards,
Ludovic

MikeOpenHWGroup, 2024-10-28:
Hi @khandelwaltanuj, I was surprised to see this Issue closed.   Is it fixed?  I just created pull-request #2546 to fix this issue, but if it is no longer needed, let me know.

khandelwaltanuj, 2024-10-28:
Hi @MikeOpenHWGroup 

My fault, i had fixed it localy to make it work. And this is not merge yet. 

Regards
Tanuj

MikeOpenHWGroup, 2024-10-28:
OK.  You can either review/merge #2546 or submit your own PR and I can close #2546.

MikeOpenHWGroup, 2024-10-29:
Resolved by #2546.

closed 2024-10-29 by MikeOpenHWGroup