openhwfoundation/core-v-verif #2545: [CV32E40S] DSIM uvmt fails to build

Northeus, 2024-10-27:
Hello,
I have some problems with running the CV32E40S UVMT testbench for hello-world test. When I try to run this testbench for CV32E40S, they result in error `FileNotFound`. Testbench for CV32E40P work normally. 

The error from the logs is:
```
...
=N:[DisabledSvaFeature]:
    The following SVA features are being disabled:    

    /home/jan/core-v-verif/core-v-cores/cv32e40s/sva/cv32e40s_alignment_buffer_sva.sv:81:3      Properties                   
    Included from /home/jan/core-v-verif/core-v-cores/cv32e40s/bhv/cv32e40s_wrapper.sv:16:12                                 

    /home/jan/core-v-verif/core-v-cores/cv32e40s/sva/cv32e40s_alignment_buffer_sva.sv:86:7      Assert property statement    
    Included from /home/jan/core-v-verif/core-v-cores/cv32e40s/bhv/cv32e40s_wrapper.sv:16:12                                 

    /home/jan/core-v-verif/core-v-cores/cv32e40s/sva/cv32e40s_mpu_sva.sv:179:25                 Cover property statement     
    Included from /home/jan/core-v-verif/core-v-cores/cv32e40s/bhv/cv32e40s_wrapper.sv:31:12                                 

    /home/jan/core-v-verif/core-v-cores/cv32e40s/sva/cv32e40s_mult_sva.sv:98:3                  Sequences                    
    Included from /home/jan/core-v-verif/core-v-cores/cv32e40s/bhv/cv32e40s_wrapper.sv:32:12                                 

=E:[FileNotFound]:
    The following files specified on the command line could not be opened:    

        

        

        

    Simulator invoked from directory: /home/jan/core-v-verif/cv32e40s/sim/uvmt    

  Metrics DSim version: 20240923.0.0 (b:R #c:0 h:e981d301ee os:ubuntu_22.04)
make: *** [/home/jan/core-v-verif/mk/uvmt/dsim.mk:137: comp] Error 1
```

I am currently at the head of the `cv32e40s/release` branch. Also, there seemed to be some issues related to DSIM from 2022. Are those issues still relevant, thus DSIM is not suitable for running testbenches on this core?

Northeus, 2024-10-28:
Thanks for a quick response! Here are the commands i used to run both CV32E40P and CV32E40S tests:
```sh
git clone git@github.com:openhwgroup/core-v-verif.git
python3.12 -m venv venv
source venv/bin/activate
source env.sh
cd /home/jan/tmp/data/core-v-verif/bin
pip3 install -r requirements.txt
cd /home/jan/tmp/data/core-v-verif/cv32e40p/sim/uvmt/
make test TEST=hello-world USE_ISS=NO
git fetch --all
git checkout cv32e40s/release
cd /home/jan/tmp/data/core-v-verif/cv32e40s/sim/uvmt/
make test TEST=hello-world USE_ISS=NO
```

Where the content of `env.sh` is:
```sh
export CORE_V_VERIF=/home/jan/tmp/data/core-v-verif
export CV_SW_TOOLCHAIN=/home/jan/programs/core-v/riscv
export CV_SW_PREFIX=riscv32-unknown-elf-
export CV_SW_MARCH=rv32imc_zicsr
export CV_SIMULATOR=dsim

export DSIM_HOME=/home/jan/metrics-ca/dsim/20240923.0.0
export STD_LIBS=/home/jan/metrics-ca/dsim/20240923.0.0/std_pkgs/lib
export LLVM_HOME=/home/jan/metrics-ca/dsim/20240923.0.0/llvm_small
export PATH=${LLVM_HOME}/bin:${DSIM_HOME}/bin:$PATH
export LD_LIBRARY_PATH=${LLVM_HOME}/lib:${DSIM_HOME}/lib
export DSIM_LICENSE=/home/jan/metrics-ca/dsim-license.json

export UVM_HOME=/home/jan/metrics-ca/dsim/20240923.0.0/uvm/1.2
```

Northeus, 2024-10-28:
Have just tried also the latest version of DSIM  (20240923.1.0), but the results are the same). The version i previously used was installed before 25.10., so it should be the older one.

MikeOpenHWGroup, 2024-10-29:
It seems there are two problems here:
1. DSim does not appear to handle some SVA code constructs properly.  I have found a work-around for this.
2. The DSim Makefile (mk/uvmt/dsim.mk) has suffered some bit-rot and does not handle the new scheme for by-passing the ImperasDV reference model (USE_ISS=NO).  I have found the problem, but not yet a solution.
 
Working on it...