openhwfoundation/core-v-verif #2561: Connection between rvfi Agent and isacov Agent in cv32e40x UVM environment

franAyachi, 2024-12-05:
Hi! I dont really know if I am missing something but in the UVM environment of cv32e40x core: 
https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40x/env/uvme/uvme_cv32e40x_env.sv

The function that connect the rvfi Agent with the isacov Agent:
```
`function void uvme_cv32e40x_env_c::connect_coverage_model();

   isacov_agent.monitor.ap.connect(cov_model.exceptions_covg.isacov_mon_export);
   isacov_agent.monitor.ap.connect(cov_model.counters_covg.isacov_mon_export);

   obi_memory_data_agent.mon_ap.connect(pma_agent.monitor.obi_d_export);
   foreach (rvfi_agent.instr_mon_ap[i]) begin
      rvfi_agent.instr_mon_ap[i].connect(isacov_agent.monitor.rvfi_instr_export);
      rvfi_agent.instr_mon_ap[i].connect(cov_model.interrupt_covg.interrupt_mon_export);
      rvfi_agent.instr_mon_ap[i].connect(pma_agent.monitor.rvfi_instr_export);
   end

endfunction: connect_coverage_model`
```
First thing I miss its the following definition: `rvfi_agent.instr_mon_ap[i]`, I dont find any monitor or port called intstr_mon_ap[], but there is a monitor created in rvfi Agent called `instr_monitor` and inside this monitor an `uvm_analysis_port` called `ap`. 

Also when connecting with isacov, the `rvfi_instr_export` is already in the Agent, not in the cov_monitor, I guess the idea would be to recieve the sequence from rvfi agent into cov agent, wich its the `rvfi_instr_export` and then, its the cov Agent who connect this port with the one defined in the cov monitor `rvfi_instr_import`.

Having said that, should not be this line **_rvfi_agent.instr_mon_ap[i].connect(isacov_agent.monitor.rvfi_instr_export);_** converted into:
**_rvfi_agent.instr_mon.ap[i].connect(isacov_agent.rvfi_instr_export);_**  ??

Thanks!!

MikeOpenHWGroup, 2024-12-05:
Hi @franAyachi, thanks for your interest in OpenHW.  The first thing to keep in mind about working with the verification environment for the CV32E40X is to use the `cv32e40x/dev` branch of this repo.  The CV32E40X has not yet had a formal v1.0.0 release, so the verification environment is still on the 'dev' branch.

> I dont find any monitor or port called intstr_mon_ap[]

Most of the UVM Agents used in the core-v-verif environments are in the [lib/uvm_agents](https://github.com/openhwgroup/core-v-verif/tree/cv32e40x/dev/lib/uvm_agents) directory.  Have a look at the [RVFI Agent](https://github.com/openhwgroup/core-v-verif/blob/76311ecd92a6d0e26a9f5539fec2573f82214308/lib/uvm_agents/uvma_rvfi/uvma_rvfi_agent.sv#L39)

franAyachi, 2024-12-05:
Okay, now all make sense.
I have review the repo and all the ports are coherent. Thanks!

closed 2024-12-05 by franAyachi