openhwfoundation/core-v-verif #2567: DUT (core) parameters about FPU

murattokez, 2024-12-12:
Hello,

I want to run riscv-dv tests in core-v-verif. in yaml file 
```
compile_flags:
         +define+NO_PULP +define+FPU
```

But then I saw the following lines in the uvmt_cv32e40p_tb.sv file.

[image]

Can't we use FPU without using PULP. What exactly does PULP enable.

MikeOpenHWGroup, 2024-12-12:
Hi @murattokez, if you are planning on using the PULP and /or F ISAs, it is recommended you use the `cv32e40p/dev` branch of this repository.  It _is_ technically possible to use the floating point unit without PULP, but the latest version of CV32E40P (v1.8.3) supports both and the verification environment enables them both.

The PULP parameter enables the PULP ISA.  Details can be found in the CORE-V Instruction Set Custom Extensions chapter of the [CV32E40P User Manual](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/instruction_set_extensions.html).

Also, if you are using `riscv-dv` within CORE-V-VERIF, please have a look at [WorkingWithCOREV-DV](https://docs.google.com/presentation/d/1hFoXlYw2vqwnVcmSE9j5cfiBGsz3PR6M/edit?usp=sharing&ouid=112401338207375688045&rtpof=true&sd=true).  This document explains how riscv-dv is used in our verification environment.

murattokez, 2024-12-12:
I am using CV32E40P (v1.8.3). Using this version I can use FPU without using PULP ISA, right? So there is no problem if I do the following update?

[image]

MikeOpenHWGroup, 2024-12-12:
Yes, that _should_ work.  Note that we have never tested this configuration, so you may encounter some unexpected issues.  If you do, please create an Issue for it.