openhwfoundation/core-v-verif #2568: [QUESTION] : Booting Linux in this modified version of Spike

jason23g, 2024-12-12:
Hello guys,

I have been trying for a couple of weeks to boot Linux(kernel version 5.10.229) in this modified version of Spike. However, it doesn't work, even if in the original repo of Spike, I can successfully boot Linux using the same fw_payload.elf and dtb files. In this modified version, the boot process reaches before the "clock switch riscv clock source" and then hangs until I stop ( approximately 19 hours). After observing the logs of these two spike simulators, I understand that there are a few differences that I haven't yet achieved to understand why. As a result, I am wondering if it is possible to boot Linux in this modified version of the spike.

Thanks in advance for your time.

MikeOpenHWGroup, 2024-12-12:
Hi @jason23g, thanks for your interest in CORE-V-VERIF.

When you say, "_this_ modified version of Spike", what are you referring to exactly?  Do you mean the .so generated from [core-v-verif/vendor/riscv/riscv-isa-sim](https://github.com/openhwgroup/core-v-verif/tree/master/vendor/riscv/riscv-isa-sim) in this repo?   To be honest, I do not know - it is certainly not our goal to do so.  In this repo, the Spike model is used as a reference model for RTL models of RISC-V cores which are running in a bare-metal environment designed to verify the RTL.

closed 2024-12-12 by MikeOpenHWGroup

jason23g, 2024-12-12:
Hi, @MikeOpenHWGroup thanks for your quick answer,

With the term "modified version of spike," I refer to the current sources of spike located in [core-v-verif/vendor/riscv/riscv-isa-sim](https://github.com/openhwgroup/core-v-verif/tree/master/vendor/riscv/riscv-isa-sim), which, from what I understand, have been modified to support the spike tandem mode in the co-simulation. In addition, would it be relatively easy for someone to support spike tandem co-simulation with the latest spike version?

Thanks in advance for your time.

MikeOpenHWGroup, 2024-12-12:
> In addition, would it be relatively easy for someone to support spike tandem co-simulation with the latest spike version?

Yes, we plan to regularly update to the latest version of Spike.

jason23g, 2024-12-13:
Hello @MikeOpenHWGroup, 

Could you tell me which commit of spike you fork when you started working with spike to support  tandem simulation.

Thanks in advance for your time.

MikeOpenHWGroup, 2024-12-13:
That is determined on a per-project basis as each project has core-specific requirements.  For example, the CVA6 is currently cloning [riscv-isa-sim hash fcbdbe](https://github.com/openhwgroup/core-v-verif/blame/6c1e999a97f61b4b805603ba142a1347f785fff2/vendor/riscv_riscv-isa-sim.vendor.hjson#L13).  I just looked and the CV32E20 is also using the same version (probably by coincidence - there is no reason to assume they will use the same version all the time).