openhwfoundation/core-v-verif #2571: Running the testbench with VCS can’t find file vcs.tcl

hjy-ucas, 2024-12-26:
## Bug Title
Running the testbench with VCS can’t find file vcs.tcl

### Type
* Other.

### Steps to Reproduce
 Command line :
 `make vcs-run`
Result：
`couldn't open ../tools/vcs/vcs_batch.tcl : no such file or directory`

[image]

### Additional context
```
cd $CORE_V_VERIF/cv32e40p/sim/core 
make
```
[image]
The solution is to add  `--Wno-MULTIDRIVEN --Wno-COMBDLY` to the Makefile.
[image]

hjy-ucas, 2024-12-26:
I have supplemented the files and content code, and it seems that the simulation can be carried out normally.
[image]

MikeOpenHWGroup, 2025-01-02:
Nice!   Would you be willing to submit a pull-request for this?   You would need to sign the [Eclipse Contributor Agreement](https://www.eclipse.org/legal/eca/) - don't worry it is painless and easy.

hjy-ucas, 2025-01-06:
> 太棒了！你愿意为此提交一个拉取请求吗？你需要签署[Eclipse 贡献者协议](https://www.eclipse.org/legal/eca/)- 别担心，这很轻松，很简单。

Thank you very much. I have already signed the Eclipse Contributor Agreement. What else do I need to do next?