openhwfoundation/core-v-verif #2577: RVFI agent: logging csr register updates with rvfi

murattokez, 2025-01-17:
In my current uvm_test_top.env.rvfi_agent.trn.log file I get the following output:

[image]

I can't see the updated csr register and its value while csr instructions in log. Is there any way to log this information?

MikeOpenHWGroup, 2025-01-17:
We do not (yet) support logging of the CSR updates.  This has long been on our ToDo list, and a PR to add it would be welcome.

MenoosGit, 2025-02-26:
@MikeOpenHWGroup and @murattokez   Can you please elaborate what is missing, I would like to work on this issue.

MikeOpenHWGroup, 2026-02-01:
> Is using the CSR fields from uvma_rvfi_instr_seq_item (rdata, wdata/wmask) the intended approach for logging CSR updates?

Yes, it is.  As you say, there should not be any need to modify the RTL or the RVFI.   Thanks for drilling down on this.

MikeOpenHWGroup, 2026-11-16:
Hi @ankit-cybertron, apologies for the delay.  I've merged your PR, please close this Issue if that resolves this one.   Thanks!