openhwfoundation/core-v-verif #2584: Followed the quick start for cv32e40p but got hit with 7 warnings

dave-roo, 2025-02-06:
*https://docs.openhwgroup.org/projects/core-v-verif/en/latest/quick_start.html

I just ran the quick-start and was it with the following error:

%Warning-COMBDLY: /home/vboxuser/Documents/core-v-verif/core-v-cores/cv32e40p/rtl/../bhv/cv32e40p_sim_clock_gate.sv:25:31: Non-blocking assignment '<=' in combinational logic process
                                                                                                                         : ... This will be executed as a blocking assignment '='!
   25 |     if (clk_i == 1'b0) clk_en <= en_i | scan_cg_en_i;
      |                               ^~
                  *** See https://verilator.org/warn/COMBDLY before disabling this,
                  else you may end up with different sim results.
%Error: Exiting due to 7 warning(s)
make[1]: *** [Makefile:491: testbench_verilator] Error 1
make[1]: Leaving directory '/home/vboxuser/Documents/core-v-verif/cv32e40p/sim/core'
make: *** [Makefile:485: sanity-veri-run] Error 2

MikeOpenHWGroup, 2025-02-07:
Hi @dave-roo, can you provide the _exact_ command-lines you used to get to this error?  Please start with `git clone ....` and include all commands up to, and including, `make ...`.

Also, a the CV32E40P is still under development, please do this from the `cv32e40p/dev` branch of core-v-verif.

dave-roo, 2025-02-09:
Hi,

(added the expected variables in bashrc)

git clone https://github.com/openhwgroup/core-v-verif.git
git switch origin/cv32e40p/dev
cd cv32e40p/sim/core
make

I have made it passed these errors by suppressing the warnings

I did suppress the warnings by adding the following to my Makefile.
-Wno-COMBDLY
-Wno-MULTIDRIVEN

and the script continued as expected.

Verilator 5.020-1

MikeOpenHWGroup, 2025-02-10:
> git clone https://github.com/openhwgroup/core-v-verif.git
git switch origin/cv32e40p/dev
cd cv32e40p/sim/core
make

That doesn't work, even for me.  😆 Try this:
$ git clone -b cv32e40p/dev https://github.com/openhwgroup/core-v-verif.git
$ cd core-v-verif/cv32e40p/sim/core
$ make

dave-roo, 2025-02-10:
I followed your steps and got the same warning messages as before. The log is attached.

[verif_corev_log_make_error.txt](https://github.com/user-attachments/files/18730507/verif_corev_log_make_error.txt)

MikeOpenHWGroup, 2025-02-10:
Well.  _That_ was fun. Verilator is not often used for the CV32E40P and so the Makefile has suffered from bit-rot. I had to find [this interesting tidbit](https://github.com/verilator/verilator/issues/4898) to fix the issue.  Since that needed fixing, I took the opportunity to clean up a few items for both Verilator and DSim, see pull-request #2585.

Let me know if that resolves the issue for you and I'll merge in the fix.  FWIW, my Verilator version is v5.032 and my DSim version is 20240923.9.0.

PR #2585 (2025-04-01): Restore Verilator compile/simulation
Resolves #2584, and cleans up DSim compiles as well.
closed 2025-03-26 by MikeOpenHWGroup