openhwfoundation/core-v-verif #2588: VCS Compiling problem Corev-dv

murattokez, 2025-03-05:
Hello,

When compiling corev-dv with VCS I get errors in 3 related files. I was able to compile with Questa. I am working on cv32e40p/dev branch. Can anyone help?

```
Error-[ICTA] Incompatible complex type
/verif/code/env/instr_gen_env/cv32e40p_isg_env/cv32e40p_load_store_instr_lib.sv, 45
  Incompatible complex type assignment
  Type of source expression is incompatible with type of target expression. 
  Mismatching types cannot be used in assignments, initializations and 
  instantiations. The type of the target is 'enum 
  riscv_instr_pkg::riscv_reg_t$[]', while the type of the source is 'enum 
  riscv_instr_pkg::riscv_reg_t$[$]'.
  Source Expression: this.s0_a5_avail_regs.find(item) with ((!(item inside 
  {this.cfg.reserved_regs, this.reserved_rd})))

Error-[ICTA] Incompatible complex type
/verif/code/env/instr_gen_env/cv32e40p_isg_env/cv32e40p_load_store_instr_lib.sv, 77
  Incompatible complex type assignment
  Type of source expression is incompatible with type of target expression. 
  Mismatching types cannot be used in assignments, initializations and 
  instantiations. The type of the target is 'enum 
  riscv_instr_pkg::riscv_reg_t$[]', while the type of the source is 'enum 
  riscv_instr_pkg::riscv_reg_t$[$]'.
  Source Expression: this.s0_a5_avail_regs.find(item) with ((!(item inside 
  {this.cfg.reserved_regs, this.reserved_rd})))
  

Error-[ICTA] Incompatible complex type
/verif/code/env/instr_gen_env/cv32e40p_isg_env/cv32e40p_loop_instr.sv, 53
  Incompatible complex type assignment
  Type of source expression is incompatible with type of target expression. 
  Mismatching types cannot be used in assignments, initializations and 
  instantiations. The type of the target is 'enum 
  riscv_instr_pkg::riscv_reg_t$[]', while the type of the source is 'enum 
  riscv_instr_pkg::riscv_reg_t$[$]'.
  Source Expression: this.s0_a5_avail_regs.find(item) with ((!(item inside 
  {this.cfg.reserved_regs, this.reserved_rd})))
```
  

135 warnings
3 errors

MikeOpenHWGroup, 2025-03-05:
This looks similar to #2393.  The root cause of that issue is a bug in older versions of VCS. For a long time we had a kludge in place as a workaround. It was eventually removed in #2402 and the conclusion was to use a newer version of VCS such as `U-2023.03-SP1` or `V-2023.12-1` (although these are no long "new" versions).

What version of VCS are you using?

murattokez, 2025-03-06:
I am using V-2023.12-SP1-1. I solved the problem by assigning the dynamic array to a temp queue. It is now working in Questa and VCS.

MikeOpenHWGroup, 2025-03-06:
It is odd that we do not see this error since we are using _almost_ the same version of VCS.  Would you be willing to contribute a PR so we can review your fix?

murattokez, 2025-03-07:
Sure, I have created PR with my fix so you can review it.