openhwfoundation/core-v-verif #2594: Missing compressed load double (c_ld) instruction

jason23g, 2025-03-15:
Hello guys,

I noticed something strange regarding the compressed load double (c_ld) instruction. In the folder riscv/insn, the c_ld.h file is missing; however, it is in the encoding.h file, there is a declaration of it: https://github.com/riscv-software-src/riscv-isa-sim/blob/3f79e3b7ded80d9ef0e722126b3765207e010711/riscv/encoding.h#L3143C3-L3144C42 . Is c_ld instruction implemented somewhere else or not at all? What should I do to contain it in the spike if it doesn't?

Thanks a lot for your time.

MikeOpenHWGroup, 2025-03-15:
Hi @jason23g, thanks for your question.

I suspect you are asking about the _lack_ of a file `core-v-verif/vendor/riscv/riscv-isa-sim/riscv/insns/c_ld.h`.  Is that correct?  If so, I suspect the answer is that currently, we are executing a full "industrial grade" verification cycle on an RV32 variants and the `c.ld` is an RV64IC instruction.  There are plans to launch a verification cycle of a 64-bit variant of CVA6 this year, so clearly we'll need to address this at that time.

@cathales, can you comment?

jason23g, 2025-03-17:
Ok could you guide me on how to add c_ld in the current implemantation of core-v-verif ?

MikeOpenHWGroup, 2025-03-17:
> Ok could you guide me on how to add c_ld in the current implemantation of core-v-verif ?

Oh, I _love_ to be able to do that. Unfortunately, no such documentation exists.  RVI has yet to produce a User Manual and/or Developer Manual for Spike (riscv-isa-sim) and at OpenHW we have maintained that tradition.  It is on my infinitely long and growing ToDo list.  You'll need to look at the examples in our repo and reverse engineer it.

cathales, 2025-03-20:
It seems to be more a verification issue than a RTL one.
Anyway, here is the expected path of a `c.ld` instruction in the pipeline.

It is decompressed in `compressed_decoder`.

https://github.com/openhwgroup/cva6/blob/b258d278167d0c6350e2b8d180e84936ded60a20/core/compressed_decoder.sv#L111-L129

The decompressed instruction is then decoded in `decoder`.

https://github.com/openhwgroup/cva6/blob/b258d278167d0c6350e2b8d180e84936ded60a20/core/decoder.sv#L1059

The `load_store_unit` checks the alignment.

https://github.com/openhwgroup/cva6/blob/b258d278167d0c6350e2b8d180e84936ded60a20/core/load_store_unit.sv#L714

The `load_unit` gets the size to load.

https://github.com/openhwgroup/cva6/blob/b258d278167d0c6350e2b8d180e84936ded60a20/core/load_unit.sv#L245

The size to load is calculated by the `extract_transfer_size` function in `ariane_pkg`.

https://github.com/openhwgroup/cva6/blob/b258d278167d0c6350e2b8d180e84936ded60a20/core/include/ariane_pkg.sv#L735

It seems everything is here for `c.ld` to work, it still has to be tested.

NOTE: the check in `load_store_unit` seems to be a repetition of the `extract_transfer_size` and should probably be replaced by a call to this function.