openhwfoundation/core-v-verif #2598: VCS Compiling problem for cv32e40p and PULP

yingjiezero, 2025-03-21:
# Issue Description

Hello, I encountered an issue while running the `cv32e40p` project. When I attempt to execute the following command in the `cv32e40p/sim/core` directory:

```bash
make vcs-custom TEST=pulp_general_alu CUSTOM=. CV_SIMULATOR=vcs
```

The simulation run fails, producing multiple errors, including a **UVM_FATAL** log entry. Below is part of the error output:

```log
ucli% run
[cv32e40p_core]: PULP_XPULP =           0, PULP_CLUSTER =           0, FPU           0, PULP_ZFINX           0, NUM_MHPMCOUNTERS           1
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(265) @ 0ns: reporter [RNDSTALL] INSTR OBI stall enable: 1
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(266) @ 0ns: reporter [RNDSTALL] INSTR OBI stall mode:   1
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(267) @ 0ns: reporter [RNDSTALL] INSTR OBI stall gnt:    1
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(268) @ 0ns: reporter [RNDSTALL] INSTR OBI stall valid:  2
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(269) @ 0ns: reporter [RNDSTALL] INSTR OBI stall max:    3
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(270) @ 0ns: reporter [RNDSTALL] DATA  OBI stall enable: 0
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(271) @ 0ns: reporter [RNDSTALL] DATA  OBI stall mode:   0
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(272) @ 0ns: reporter [RNDSTALL] DATA  OBI stall gnt:    0
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(273) @ 0ns: reporter [RNDSTALL] DATA  OBI stall valid:  0
UVM_INFO /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(274) @ 0ns: reporter [RNDSTALL] DATA  OBI stall max:    0
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 6140ns: reporter [MM_RAM] out of bounds read from ccda43ac (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 15980ns: reporter [MM_RAM] out of bounds read from 0550107b (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 15980ns: reporter [MM_RAM] out of bounds read from 0550107c (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 15990ns: reporter [MM_RAM] out of bounds read from 0550107c (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 15990ns: reporter [MM_RAM] out of bounds read from 0550107f (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 16080ns: reporter [MM_RAM] out of bounds read from 055010d3 (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 16080ns: reporter [MM_RAM] out of bounds read from 055010d4 (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 16090ns: reporter [MM_RAM] out of bounds read from 055010d4 (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
UVM_ERROR /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(487) @ 16090ns: reporter [MM_RAM] out of bounds read from 055010d7 (RAM_ADDR_WIDTH=22; dm_halt_addri=1a110800, DBG_ADDR_WIDTH=14)
"/home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv", 445: tb_top.cv32e40p_tb_wrapper_i.ram_i.out_of_bounds_write: started at 16120000ps failed at 16120000ps
        Offending '(((((((((((((data_addr_i < (2 ** RAM_ADDR_WIDTH)) || ((data_addr_i >= dm_halt_addr_i) && (data_addr_i < (dm_halt_addr_i + (2 ** DBG_ADDR_WIDTH))))) || (data_addr_i == MMADDR_PRINT)) || (data_addr_i == MMADDR_TIMERREG)) || (data_addr_i == MMADDR_TIMERVAL)) || (data_addr_i == MMADDR_DBG)) || (data_addr_i == MMADDR_TESTSTATUS)) || (data_addr_i == MMADDR_EXIT)) || (data_addr_i == MMADDR_SIGBEGIN)) || (data_addr_i == MMADDR_SIGEND)) || (data_addr_i == MMADDR_SIGDUMP)) || (data_addr_i == MMADDR_TICKS)) || (data_addr_i[31:16] == MMADDR_RNDSTALL))'
UVM_FATAL /home/gk/Desktop/Package/gk/core-v-verif/cv32e40p/tb/core/mm_ram.sv(462) @ 16120ns: reporter [MM_RAM] out of bounds write to 0550107c with 00000020

--- UVM Report Summary ---

** Report counts by severity
UVM_INFO :   10
UVM_WARNING :    0
UVM_ERROR :    9
UVM_FATAL :    1
** Report counts by id
[MM_RAM]    10
[RNDSTALL]    10
$finish called from file "/home/gk/Desktop/Package/vcs2018/vcs/O-2018.09-SP2/etc/uvm/base/uvm_report_object.svh", line 292.
```
---

### Changes I Made

To enable PULP functionality and try to resolve the issue, I made the following changes:

1. Modified the `PULP_XPULP` parameter in `cv32e40p_core.sv`:
    ```verilog
    parameter PULP_XPULP = 1;
    ```
2. Uncommented the following lines (327–329) in `Common.mk`:
    ```makefile
    ifeq ($(firstword $(subst _, ,$(TEST))),pulp)
    CFLAGS = -Os -g -D__riscv__=1 -D__LITTLE_ENDIAN__=1 -march=rv32imcxpulpv2 -Wa,-march=rv32imcxpulpv2 -fdata-sections -ffunction-sections -fdiagnostics-color=always
    endif
    ```
3. Configured toolchain environment variables for PULP GCC:
    ```bash
    export CV_SW_TOOLCHAIN=/home/gk/Desktop/Package/v1.0.16-pulp-riscv-gcc-ubuntu-16
    export CV_SW_PREFIX=riscv32-unknown-elf-
    export CV_SW_MARCH=rv32gc
    ```
---

# Request for Help

I am seeking guidance on how to properly configure the project to enable support for PULP-based tests (`TEST=pulp_general_alu`). A similar setup with **corev-openhw-gcc-ubuntu2004-20241020** as the toolchain was able to successfully run `TEST=hello-world`, but it also failed when running `TEST=pulp_general_alu`.

---

**Additional Information**
Host system: Ubuntu 20.04
VCS version: O-2018.09-SP2_Full64
Hardware compiler: PULP GCC v1.0.16 riscv

---

# Attachment Details
I have attached a file named full_terminal_output.log to this issue. The file contains the complete terminal logs that resulted from running the failing command. Please review this file for further insights into the error.

[full_terminal_output.log](https://github.com/user-attachments/files/19392982/full_terminal_output.log)

MikeOpenHWGroup, 2025-03-21:
Hi @yingjiezero, thanks for your interest in OpenHW and core-v-verif. And a double-thank-you for providing such a detailed Issue report.  It really helps!

I can see three problems here:
1.  You are attempting to run pulp test-programs on the `core` testbench.  This is not supported and would require a substantial update to the memory model (MM_RAM).  If you want to run the pulp test-programs, you should be using the UVM environment.
2. Your version of VCS is very old and there may compatibility issues, especially with the UVM environment and the functional coverage code (see Issue #2404 for an example).
3. We do not use the PULP toolchain.  I strongly recommend using one of the [prebuilt toolchains generated by Embecosm](https://embecosm.com/downloads/tool-chain-downloads/#pulp-top-of-tree-compilers).

yingjiezero, 2025-03-21:
Thank you for your guidance and insights. I have made some changes based on your recommendations and have attempted to run the tests in a different setup. However, I'm encountering a new set of errors. Here's a summary of what I've done:

### Current Attempt

I am now running the tests in the `cv32e40p/sim/uvmt` directory with the following command:

```bash
make test TEST=pulp_general_alu CV_SIMULATOR=vcs USE_ISS=NO COV=YES
```

### New Error Details

While the setup is slightly different, the simulation still results in errors, but with different messages:

```
UVM_INFO @ 16.800 ns : uvmt_cv32e40p_base_test.sv(281) uvm_test_top [BASE TEST] set load_instr_mem
UVM_INFO @ 144.300 ns : uvmt_cv32e40p_firmware_test.sv(120) uvm_test_top [TEST] Started RUN
779.000 ns: Illegal instruction (core 0) at PC 0x0000019a:
"/home/gk/Desktop/Package/gk/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv", 234: uvmt_cv32e40p_tb.dut_wrap.obi_data_memory_assert_i.u_assert.a_addr_be_consistent: started at 2100300ps failed at 2100300ps
	Offending '(get_min_be(addr) <= be)'
UVM_ERROR @ 2100.300 ns : uvma_obi_memory_assert.sv(236) reporter [OBIMEMASRT] be of 0x1 not consistent with addr 0x0550107c
"/home/gk/Desktop/Package/gk/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv", 234: uvmt_cv32e40p_tb.dut_wrap.obi_data_memory_assert_i.u_assert.a_addr_be_consistent: started at 2118300ps failed at 2118300ps
	Offending '(get_min_be(addr) <= be)'
UVM_ERROR @ 2118.300 ns : uvma_obi_memory_assert.sv(236) reporter [OBIMEMASRT] be of 0x7 not consistent with addr 0x055010d4
"/home/gk/Desktop/Package/gk/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv", 234: uvmt_cv32e40p_tb.dut_wrap.obi_data_memory_assert_i.u_assert.a_addr_be_consistent: started at 2127300ps failed at 2127300ps
	Offending '(get_min_be(addr) <= be)'
UVM_ERROR @ 2127.300 ns : uvma_obi_memory_assert.sv(236) reporter [OBIMEMASRT] be of 0x1 not consistent with addr 0x0550107c
"/home/gk/Desktop/Package/gk/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv", 234: uvmt_cv32e40p_tb.dut_wrap.obi_data_memory_assert_i.u_assert.a_addr_be_consistent: started at 2142300ps failed at 2142300ps
	Offending '(get_min_be(addr) <= be)'
UVM_ERROR @ 2142.300 ns : uvma_obi_memory_assert.sv(236) reporter [OBIMEMASRT] be of 0x7 not consistent with addr 0x055010d4
"/home/gk/Desktop/Package/gk/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv", 234: uvmt_cv32e40p_tb.dut_wrap.obi_data_memory_assert_i.u_assert.a_addr_be_consistent: started at 2175300ps failed at 2175300ps
	Offending '(get_min_be(addr) <= be)'
UVM_ERROR @ 2175.300 ns : uvma_obi_memory_assert.sv(236) reporter [OBIMEMASRT] be of 0x1 not consistent with addr 0x0550107c
UVM_INFO @ 2175.300 ns : uvm_report_server.svh(894) reporter [UVM/REPORT/SERVER] 
--- UVM Report Summary ---

Quit count reached!
Quit count :     5 of     5
** Report counts by severity
UVM_INFO :   50
UVM_WARNING :    0
UVM_ERROR :    5
UVM_FATAL :    0
** Report counts by id
[BASE TEST]     4
[CLKNRST]     4
[CORE_CNTRL_IF]     1
[DEBUGCOVG]     1
[INTERRUPTCOVG]     1
[MEMREADMEMH]     1
[OBIMEMASRT]     5
[OBIVPSEQ]    22
[OBI_MEMORY_MON]     4
[OVPWRAP]     1
[RNTST]     1
[RST_VSEQ]     3
[Step-andCompare]     1
[TEST]     2
[TEST_CFG]     1
[UVM/RELNOTES]     1
[UVME_CV32E40P_ENV]     1
[rv32isa_covg]     1

$finish called from file "/home/gk/Desktop/Package/vcs2018/vcs/O-2018.09-SP2/etc/uvm-1.2/base/uvm_root.svh", line 135.

uvmt_cv32e40p_tb.end_of_test: *** Test Summary ***

    FFFFFFFF   AAAAAA   IIIIII  LL        EEEEEEEE  DDDDDDD       
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FFFFF     AAAAAAAA    II    LL        EEEEE     DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA  IIIIII  LLLLLLLL  EEEEEEEE  DDDDDDD       
    --------------------------------------------------------
                   SIMULATION FAILED - ABORTED              
    --------------------------------------------------------
$finish at simulation time 2175.300 ns
```

### Changes Made

1. **Directory Change**: I have shifted to running tests in the `cv32e40p/sim/uvmt` directory as per your suggestion to use the UVM environment.
2. **Toolchain Update**: Switched to using the recommended GCC toolchain:
    ```bash
    export CV_SW_TOOLCHAIN=/home/gk/Desktop/Package/pulp-gcc-ubuntu2004-20200913
    export CV_SW_PREFIX=riscv32-unknown-elf-
    export CV_SW_MARCH=rv32gc
    ```
3. **VCS Version**: Although I attempted to install newer versions of VCS (2024 and 2022), they were not compatible with my Ubuntu 2004 setup, resulting in failures to run. Therefore, I am currently using VCS version 2018 to see if it can work.

I have attached a file named make_test.log to this issue. The file contains the complete terminal logs that resulted from running the failing command. Please review this file for further insights into the error.

[make_test.log](https://github.com/user-attachments/files/19394301/make_test.log)

MikeOpenHWGroup, 2025-03-22:
Ahahaha!  You have encountered a bug that we just recently discovered as well. (Embarrassing!)  If you are sign-up for an [Eclipse account](https://www.eclipse.org/), you can see the discussion about this on our MatterMost chat-board ([link](https://mattermost.openhwgroup.org/all-users/pl/jpidc4kb9trjifzoomj965gn8a)).  We will create a GitHub issue about this soon.

The assertion that is firing is in [uvma_obi_memory_assert.sv](https://github.com/openhwgroup/core-v-verif/blob/44bbfbfb01dc28c3ea6c2e6d2e4bc86c0c887f9e/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv#L231).  You can do one or more of these things:
1. Comment out the assertion.  We will be fixing it sometime next week.
2. Patch the property to this:
```
  property p_addr_be_consistent;
    disable iff (DATA_WIDTH !== 32) req |-> addr[1:0] <= get_addr_lsb(be);  // is that right ??
  endproperty : p_addr_be_consistent
```
3. Use the `cv32e40p/dev` branch of this repo.  This is _strongly recommended_ as this is the most up-to-date version of the repo for the CV32E40P core.

Lastly, you should not use `rv32gc` for your **CV_SW_MARCH** shell env variable.  The CV32E40P does _not_ support the rv32gc profile.  The CV32E40P test-program environment configuration defaults this var to `rv32imc_zicsr_zifencei`, so you can either unset your CV_SW_MARCH, or use `rv32imc_zicsr_zifencei`.  If you want to learn about how this all works in core-v-verif, please read the [TOOLCHAIN](https://github.com/openhwgroup/core-v-verif/blob/master/mk/TOOLCHAIN.md) documentation.

yingjiezero, 2025-03-23:
Thank you for your previous guidance and recommendations. I have made several adjustments based on your input and attempted to run the tests, but I am still encountering errors with the `make test TEST=pulp_general_alu CV_SIMULATOR=vcs USE_ISS=NO COV=YES` command. Below are the details of the current issues and the changes I have made:

---

### Current Errors

1. **While compiling `uvme_cv32e40p_fp_instr_covg.sv`:**
   ```
   Error-[FCIWCS] Invalid 'with' clause specification
   /home/gk/Desktop/Package/core-v-verif/cv32e40p/env/uvme/cov/uvme_cv32e40p_fp_instr_covg.sv, 396
   uvme_cv32e40p_pkg, "cp_curr_fpu_inst_rd_for_0_lat_apu_result"
     The 'with' clause specified on the coverpoint bin does not have any impact 
     and is invalid; it should contain at least one occurence of the 'item' 
     keyword.
     Please remove or correct the clause and compile
   ```

2. **Errors during assembly of `pulp_general_alu.S`:**
   ```
   /home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S: Assembler messages:
   /home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:67: 错误： unrecognized opcode `cv.abs x18,x20'
   /home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:73: 错误： unrecognized opcode `cv.abs x18,x20'
   /home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:79: 错误： unrecognized opcode `cv.abs x18,x20'
   /home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:85: 错误： unrecognized opcode `cv.abs x18,x20'
   /home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:91: 错误： unrecognized opcode `cv.abs x18,x20'
   /home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:97: 错误： unrecognized opcode `cv.abs x18,x20'
······
/home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:1423: 错误： unrecognized opcode `cv.suburnr x18,x20,x21'
/home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:1431: 错误： unrecognized opcode `cv.suburnr x18,x20,x21'
/home/gk/Desktop/Package/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S:1439: 错误： unrecognized opcode `cv.suburnr x18,x20,x21'
   ```

---

### Changes Made

To align as closely as possible with the setup you suggested, I have made the following modifications:

1. **Branch Update**: I am now using the `cv32e40p/dev` branch of the repository.
2. **Updated Toolchain**: Switched to GCC toolchain with these configurations:
   ```bash
   export CV_SW_TOOLCHAIN=/home/gk/Desktop/Package/pulp-gcc-ubuntu2004-20200913
   export CV_SW_PREFIX=riscv32-unknown-elf-
   export CV_SW_MARCH=rv32imc_zicsr_zifencei
   ```
3. **VCS Version**: Upgraded to VCS version `T-2022.06-SP2-10`, as the previously used version was outdated and caused compatibility issues.

MikeOpenHWGroup, 2025-03-23:
Your compilation error is due to Issue #2404, which I referred to before.  If you look at that Issue you will see that we are recommending VCS version V-2023.12-1 or newer.  This was a bug in VCS, so we will not change our code, since it works with all other commercial SystemVerilog simulators and the newer versions of VCS have solved the problem.

As for the compiler issue, since you are overriding the CV_SW_MARCH with a shell environment variable and setting it to `rv32imc_zicsr_zifencei`, you will not be able to compile test-programs with instructions from the PULP ISA.  As stated earlier, it is recommended to unset your CV_SW_MARCH and let the CFG scripts do it for you as explained in [TOOLCHAIN](https://github.com/openhwgroup/core-v-verif/blob/master/mk/TOOLCHAIN.md).

yingjiezero, 2025-03-24:
I attempted to build the PULP toolchain from source following the instructions in TOOLCHAIN.md. However, during the `sudo make` step, I encountered compilation errors related to unresolved symbols like `isl_id_free` and others.

Here are the details of the error:
- The environment variables were set as below:
  ```bash
  export PATH=/home/gk/Desktop/Package/pulp/bin:$PATH
  export PULP="YES"
  ```
  The library versions I used were from `pulp-gcc-ubuntu2004-20200913`.
- Error Details
```
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings -fno-common  -DHAVE_CONFIG_H -I. -I. -I/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc -I/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/. -I/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/../include -I/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/../libcpp/include  -I/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/../libdecnumber -I/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/../libbacktrace   -o graphite-isl-ast-to-gimple.o -MT graphite-isl-ast-to-gimple.o -MMD -MP -MF ./.deps/graphite-isl-ast-to-gimple.TPo /home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In function ‘void ivs_params_clear(ivs_params&)’:
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:98:7: error: ‘isl_id_free’ was not declared in this scope; did you mean ‘isl_aff_free’?
   98 |       isl_id_free (it->first);
      |       ^~~~~~~~~~~
      |       isl_aff_free
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function ‘tree_node* translate_isl_ast_to_gimple::gcc_expression_from_isl_ast_expr_id(tree, isl_ast_expr*, ivs_params&)’:
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:292:3: error: ‘isl_id_free’ was not declared in this scope; did you mean ‘isl_aff_free’?
  292 |   isl_id_free (tmp_isl_id);
      |   ^~~~~~~~~~~
      |   isl_aff_free
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function ‘loop* translate_isl_ast_to_gimple::graphite_create_new_loop(edge, isl_ast_node*, loop_p, tree, tree, tree, ivs_params&)’:
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:628:5: error: ‘isl_id_free’ was not declared in this scope; did you mean ‘isl_aff_free’?
  628 |     isl_id_free (res->first);
      |     ^~~~~~~~~~~
      |     isl_aff_free
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function ‘edge_def* translate_isl_ast_to_gimple::translate_isl_ast_for_loop(loop_p, isl_ast_node*, edge, tree, tree, tree, ivs_params&)’:
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:668:53: error: ‘isl_id_get_user’ was not declared in this scope; did you mean ‘isl_arg_user’?
  668 |       ast_build_info *for_info = (ast_build_info *) isl_id_get_user (id);
      |                                                     ^~~~~~~~~~~~~~~
      |                                                     isl_arg_user
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:671:7: error: ‘isl_id_free’ was not declared in this scope; did you mean ‘isl_aff_free’?
  671 |       isl_id_free (id);
      |       ^~~~~~~~~~~
      |       isl_aff_free
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In member function ‘edge_def* translate_isl_ast_to_gimple::translate_isl_ast_node_user(isl_ast_node*, edge, ivs_params&)’:
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:875:31: error: ‘isl_id_get_user’ was not declared in this scope; did you mean ‘isl_arg_user’?
  875 |   poly_bb_p pbb = (poly_bb_p) isl_id_get_user (name_id);
      |                               ^~~~~~~~~~~~~~~
      |                               isl_arg_user
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:881:3: error: ‘isl_id_free’ was not declared in this scope; did you mean ‘isl_aff_free’?
  881 |   isl_id_free (name_id);
      |   ^~~~~~~~~~~
      |   isl_aff_free
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c: In function ‘isl_id* ast_build_before_for(isl_ast_build*, void*)’:
/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/riscv-gcc/gcc/graphite-isl-ast-to-gimple.c:2876:19: error: ‘isl_space_dim’ was not declared in this scope; did you mean ‘isl_aff_dim’?
 2876 |   int dimension = isl_space_dim (schedule_space, isl_dim_out);
      |                   ^~~~~~~~~~~~~
      |                   isl_aff_dim
[... cut ...]
make[2]: *** [Makefile:1099：graphite-isl-ast-to-gimple.o] 错误 1
make[2]: 离开目录“/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/build-gcc-newlib-stage1/gcc”
make[1]: *** [Makefile:4220：all-gcc] 错误 2
make[1]: 离开目录“/home/gk/Desktop/Package/pulp-riscv-gnu-toolchain/master/build-gcc-newlib-stage1”
make: *** [Makefile:227：stamps/build-gcc-newlib-stage1] 错误 2
```

### Attempts Made
1. Installed the latest version of `libisl-dev` to address potential compatibility issues and attempted recompilation.

MikeOpenHWGroup, 2025-03-26:
Hi @yingjiezero.  The bug in [uvma_obi_memory_assert.sv](https://github.com/openhwgroup/core-v-verif/blob/44bbfbfb01dc28c3ea6c2e6d2e4bc86c0c887f9e/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv#L231) as reported in #2599 was resolved with #2601.

yingjiezero, 2025-03-28:
I've been working on configuring the environment and running tests as discussed. Here are the steps and issues encountered:

### Setup and Configuration
I followed the recommendation to use the PULP toolchain from [here](https://github.com/pulp-platform/pulp-riscv-gnu-toolchain), specifically for `xpulpv2`. The environment is set up as follows:
```bash
export CV_SW_TOOLCHAIN=/home/host/Desktop/package/xpulpv2/riscv-pulp
export CV_SW_PREFIX=riscv32-unknown-elf-
export RISCV=/home/host/Desktop/package/xpulpv2/riscv-pulp
export PATH=$RISCV/bin:$PATH
```

### Issues Encountered
1. When using the `cv32e40p/dev` branch and running `make test TEST=pulp_general_alu CV_SIMULATOR=vcs USE_ISS=NO CFG=pulp`, an error occurs:
   ```
   cc1: error: -march=rv32imc_zicsr_zifencei_xcvhwlp_xcvmem_xcvmac_xcvbi_xcvalu_xcvsimd_xcvbitmanip: unsupported ISA substring '_ZICSR_ZIFENCEI_XCVHWLP_XCVMEM_XCVMAC_XCVBI_XCVALU_XCVSIMD_XCVBITMANIP'
   ```
   I modified `pulp.yaml` to use `-march=rv32imcxpulpv2`, but additional errors arose due to unrecognized opcodes like `cv.start`, `cv.end`, and `cv.count`.
```

2. Switching to the `master` branch and modifying `uvma_obi_memory_assert.sv` as per #2601 resulted in continuous "Illegal instruction" errors during simulation:
   ```
   Illegal instruction (core 0) at PC 0x00000002:
   ```

3. I also tried installing `xpulpv3` via the [riscv-gnu-toolchain](https://github.com/pulp-platform/riscv-gnu-toolchain) URL with similar issues persisting. However, I confirmed my `xpulpv2` setup seems sound since a simple C program utilizing PULP instructions compiles successfully:
   ```c
   #include <stdio.h>

   int main() {
       int a = 3, b = 0;
       asm volatile(
           "p.addN x0, x1, x2, 1\n"
           "p.subN x0, x1, x2, 1\n"
       );
   }
   ```

### Inquiries
- Are there known issues with the `cv32e40p` environment or the specific `pulp_general_alu` test concerning instruction set recognition?
- Is there an existing configuration or patch to resolve these opcode recognition errors in the simulation environment?
- Should I consider alternative branches or configurations that better align with the PULP ISA for these tests, especially concerning `cv.start`, `cv.end`, or related opcodes?

Thank you for any guidance or insights you can provide to help resolve these issues and allow the tests to run successfully.

Best regards,
[Your Name]

---

这种回复总结了你为解决当前问题所尝试的步骤和所遇到的具体错误，并提出具体问题，以便对方能够更准确地提供指导和帮助。

MikeOpenHWGroup, 2025-03-28:
At this point, I am not sure, so let me dig a little deeper for you.  In the meantime:
- try `CV_SW_MARCH=rv32imcxpulpv2_xcv`
- don't waste your time on the master branch.  :-)

yingjiezero, 2025-03-31:
`cv32e40p/dev branch` , try `CV_SW_MARCH=rv32imcxpulpv2_xcv`，an error occurs:
```
/home/host/Desktop/package/xpulpv2/riscv-pulp/bin/riscv32-unknown-elf-gcc \
        -DNO_PULP \
        -Os -g -static -mabi=ilp32 -march=rv32imcxpulpv2_xcv -Wall -pedantic  \
         \
        -I ../../tests/asm \
        -I /home/host/Desktop/Coverage/CV32E40P/DEV/core-v-verif/cv32e40p/bsp \
        -o /home/host/Desktop/Coverage/CV32E40P/DEV/core-v-verif/cv32e40p/sim/uvmt/vcs_results/default/pulp_general_alu/0/test_program/pulp_general_alu.elf \
        -nostartfiles \
        /home/host/Desktop/Coverage/CV32E40P/DEV/core-v-verif/cv32e40p/tests/programs/custom/pulp_general_alu/pulp_general_alu.S \
        -T /home/host/Desktop/Coverage/CV32E40P/DEV/core-v-verif/cv32e40p/bsp/link.ld \
        -Xlinker -Map=/home/host/Desktop/Coverage/CV32E40P/DEV/core-v-verif/cv32e40p/sim/uvmt/vcs_results/default/pulp_general_alu/0/test_program/pulp_general_alu.map \
         -L /home/host/Desktop/Coverage/CV32E40P/DEV/core-v-verif/cv32e40p/sim/uvmt/vcs_results/default/pulp_general_alu/0/test_program/bsp \
        -lcv-verif
cc1: error: -march=rv32imcxpulpv2_xcv: unsupported ISA substring '_XCV'
make: *** [/home/host/Desktop/Coverage/CV32E40P/DEV/core-v-verif/mk/Common.mk:525: /home/host/Desktop/Coverage/CV32E40P/DEV/core-v-verif/cv32e40p/sim/uvmt/vcs_results/default/pulp_general_alu/0/test_program/pulp_general_alu.elf] Error 1
```