openhwfoundation/core-v-verif #2599: OBI wrong assertion

AyoubJalali, 2025-03-23:
Hello,
as I discussed with @MikeOpenHWGroup in [mattermost](https://mattermost.openhwgroup.org/all-users/channels/twg--verification) about an assertion that fire in our CVA6 testbench specifically cv32a60x (as it's used obi now), [the assertion](https://github.com/openhwgroup/core-v-verif/blob/44bbfbfb01dc28c3ea6c2e6d2e4bc86c0c887f9e/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv#L232) check R-8 paragraph in [v1.2](https://raw.githubusercontent.com/openhwgroup/obi/188c87089975a59c56338949f5c187c1f8841332/OBI-v1.2.pdf), and R-9 paragraph in [v1.6 (https://github.com/openhwgroup/obi/blob/072d9173c1f2d79471d6f2a10eae59ee387d4c6f/OBI-v1.6.0.pdf). Normally the check should be for least significant addr bits (i.e. addr[1:0] in case DATA_WIDTH = 32).
I didn't get the get_min_be goal here in the property.

I'll open a PR to suggest a fix for this as so as possible

MikeOpenHWGroup, 2025-03-24:
Hi @AyoubJalali, please have a look at the attached spreadsheet. It attempts to illustrate the legal encodings of `be[3:0]` and `addr[1:0]` (according to R-8) and which bytes of a word are selected for each legal encoding. Note that due to R-7 in the OBI spec, many of the `be[3:0]` encodings are illegal.

I must admit that most of the cases where `addr[1:0]==2'b01` do not make sense to me either, but according to the spec., these seem to be legal.

[OBI_R-8.xlsx](https://github.com/user-attachments/files/19438510/OBI_R-8.xlsx)

AyoubJalali, 2025-03-26:
Fixed by #2601

closed 2025-03-26 by AyoubJalali