openhwfoundation/core-v-verif #2604: [cv_dv_utils memory partition] Always return a 32 bit address

khandelwaltanuj, 2025-03-28:
Hello,

Memory partiton divides memory in small patitions and can be used to return a random address within a choses partition. 
It uses $urandom_range(hihg_addr, low_addr) to get the adress. 
Irrespective of the address width, it always return a 32 bit address, which is not correct. 

Regards
Tanuj Khandelwal

MikeOpenHWGroup, 2025-03-28:
Hi @khandelwaltanuj, thanks for the issue.  Can you point out the specific file(s) in `cv_dv_utils` that have this bug?  Is [lib/cv_dv_utils/uvm/memory_partition/memory_partitions_cfg.svh, line 137](https://github.com/openhwgroup/core-v-verif/blob/ff68c89bd06d81767d0a4c12466db38f958c6eff/lib/cv_dv_utils/uvm/memory_partition/memory_partitions_cfg.svh#L137) is an example?

khandelwaltanuj, 2025-03-28:
Yes this is the exact bug