openhwfoundation/core-v-verif #2625: %Warning-MULTIDRIVEN: /core-v-verif/cv32e40p/tb/core/mm_ram.sv

mamed00-dotcom, 2025-06-25:
## Setting CORE-V Toolchain
```bash
export CV_SW_TOOLCHAIN="/opt/corev"
export CV_CORE=cv32e40p
export CV_SW_CFLAGS="-O2 -march=rv32imc -mabi=ilp32"
export CV_SW_CC=riscv32-corev-elf-gcc
export CV_SW_MARCH=rv32imc
export CV_SW_PREFIX=riscv32-corev-elf-
export PATH=/opt/corev/bin:$PATH

cd /core-v-verif
cd cv32e40p/sim/core
core-v-verif/cv32e40p/sim/core$ make
```
Output : 

/core-v-verif/mk/Common.mk:318: RISCV set to /opt/corev
/core-v-verif/mk/Common.mk:319: RISCV_PREFIX set to riscv32-corev-elf-
/core-v-verif/mk/Common.mk:320: RISCV_EXE_PREFIX set to /opt/corev/bin/riscv32-corev-elf-
/core-v-verif/mk/Common.mk:321: RISCV_MARCH set to rv32imc
/core-v-verif/mk/Common.mk:322: RISCV_CC set to riscv32-corev-elf-gcc
/core-v-verif/mk/Common.mk:323: RISCV_CFLAGS set to -O2 -march=rv32imc -mabi=ilp32
make veri-test TEST=hello-world
make[1]: Entering directory '/core-v-verif/cv32e40p/sim/core'
/core-v-verif/mk/Common.mk:318: RISCV set to /opt/corev
/core-v-verif/mk/Common.mk:319: RISCV_PREFIX set to riscv32-corev-elf-
/core-v-verif/mk/Common.mk:320: RISCV_EXE_PREFIX set to /opt/corev/bin/riscv32-corev-elf-
/core-v-verif/mk/Common.mk:321: RISCV_MARCH set to rv32imc
/core-v-verif/mk/Common.mk:322: RISCV_CC set to riscv32-corev-elf-gcc
/core-v-verif/mk/Common.mk:323: RISCV_CFLAGS set to -O2 -march=rv32imc -mabi=ilp32
*******************************************************************************************
* Cloning CV32E40P RTL model
*******************************************************************************************
git clone https://github.com/openhwgroup/cv32e40p /core-v-verif/core-v-cores/cv32e40p; cd /core-v-verif/core-v-cores/cv32e40p; git checkout fcd5968
fatal: destination path '/core-v-verif/core-v-cores/cv32e40p' already exists and is not an empty directory.
M	bhv/cv32e40p_sim_clock_gate.sv
HEAD is now at fcd5968 deleted redundant mul custom instructions (#661)
*******************************************************************************************
* Compiling CORE TB and CV32E40P with Verilator
*******************************************************************************************
verilator --cc --sv --exe \
	 \
	--Wno-lint --Wno-UNOPTFLAT \
	--Wno-MODDUP --top-module \
	tb_top_verilator /core-v-verif/cv32e40p/tb/core/tb_top_verilator.sv /core-v-verif/cv32e40p/tb/core/cv32e40p_tb_wrapper.sv /core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_rvalid_stall.sv /core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_gnt_stall.sv /core-v-verif/cv32e40p/tb/core/mm_ram.sv /core-v-verif/cv32e40p/tb/core/dp_ram.sv \
	-f /core-v-verif/core-v-cores/cv32e40p/cv32e40p_manifest.flist \
	/core-v-verif/core-v-cores/cv32e40p/bhv/cv32e40p_core_log.sv \
	/core-v-verif/cv32e40p/tb/core/tb_top_verilator.cpp --Mdir cobj_dir \
	-CFLAGS "-std=gnu++11 -O2" \
	-Wno-BLKANDNBLK  
%Warning-MULTIDRIVEN: /core-v-verif/cv32e40p/tb/core/mm_ram.sv:291:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_req_dec'
                                                                    : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:291:9: 
  291 |         data_req_dec        = '0;
      |         ^~~~~~~~~~~~
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:196:8: ... Location of other write
  196 |        data_req_dec = data_req_i;
      |        ^~~~~~~~~~~~
                      ... For warning description see https://verilator.org/warn/MULTIDRIVEN?v=5.008
                      ... Use "/* verilator lint_off MULTIDRIVEN */" and lint_on around source to disable this message.
%Warning-MULTIDRIVEN: /core-v-verif/cv32e40p/tb/core/mm_ram.sv:292:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_addr_dec'
                                                                    : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:292:9: 
  292 |         data_addr_dec       = '0;
      |         ^~~~~~~~~~~~~
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:205:11: ... Location of other write
  205 |           data_addr_dec  = data_addr_i[RAM_ADDR_WIDTH-1:0];
      |           ^~~~~~~~~~~~~
%Warning-MULTIDRIVEN: /core-v-verif/cv32e40p/tb/core/mm_ram.sv:293:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_wdata_dec'
                                                                    : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:293:9: 
  293 |         data_wdata_dec      = '0;
      |         ^~~~~~~~~~~~~~
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:208:8: ... Location of other write
  208 |        data_wdata_dec = data_wdata_i;
      |        ^~~~~~~~~~~~~~
%Warning-MULTIDRIVEN: /core-v-verif/cv32e40p/tb/core/mm_ram.sv:294:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_we_dec'
                                                                    : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:294:9: 
  294 |         data_we_dec         = '0;
      |         ^~~~~~~~~~~
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:209:8: ... Location of other write
  209 |        data_we_dec    = data_we_i;
      |        ^~~~~~~~~~~
%Warning-MULTIDRIVEN: /core-v-verif/cv32e40p/tb/core/mm_ram.sv:295:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_be_dec'
                                                                    : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:295:9: 
  295 |         data_be_dec         = '0;
      |         ^~~~~~~~~~~
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:210:8: ... Location of other write
  210 |        data_be_dec    = data_be_i;
      |        ^~~~~~~~~~~
%Warning-MULTIDRIVEN: /core-v-verif/cv32e40p/tb/core/mm_ram.sv:312:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'transaction'
                                                                    : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:312:9: 
  312 |         transaction         = T_PER;
      |         ^~~~~~~~~~~
                      /core-v-verif/cv32e40p/tb/core/mm_ram.sv:211:8: ... Location of other write
  211 |        transaction    = T_RAM;
      |        ^~~~~~~~~~~
%Error: Exiting due to 6 warning(s)
make[1]: *** [Makefile:491: testbench_verilator] Error 1
make[1]: Leaving directory '/core-v-verif/cv32e40p/sim/core'
make: *** [Makefile:485: sanity-veri-run] Error 2

## Verilator is failing
Verilator treats %Warning-MULTIDRIVEN as an error by default, and will not continue the build.
Should we modify `/core-v-verif/cv32e40p/tb/core/mm_ram.sv` manually ?

ps: 
```bash
verilator --version
Verilator 5.008 2023-03-04 rev v5.008
```

MikeOpenHWGroup, 2025-06-25:
Hi @mamed00-dotcom, thanks for your interest in OpenHW, and your detailed description of your issue.  Both are much appreciated.

The so-called "core" testbench for CV32E40P is rarely used and so sometimes suffers from bit-rot.  This appears to be the case here.  Verilator versions after v5.0 consider the MULTIDRIVEN lint warning as an error.  This is correct for RTL, but overly restrictive for testbench components such as the mm_ram.

There are possible two solutions:
1. Update mm_ram to remove the offending lint warnings.
2. Suppress them (in the Makefile).

I've labeled this one a "good first issue", so if you'd like to acquire instant fame and fortune, I'd be happy to review your PR to resolve this.  Please review [CONTRIBUTING](https://github.com/openhwgroup/core-v-verif/blob/master/CONTRIBUTING.md) first.

mamed00-dotcom, 2025-06-26:
Hello @MikeOpenHWGroup , thank you for your response. 
I appreciate the explanation and the guidance. Regarding the two suggested solutions (updating `mm_ram` to remove the lint warnings vs. suppressing them in the Makefile), which one do you think is more feasible or preferable for the long term?
My main goal is to be able to run the existing CSR-related scripts in CV32E40P without issues.

Thanks again for your support!