openhwfoundation/core-v-verif #2630: Suppress Verilator Warnings Causing Build Failure in Simulation

Uzair-90, 2025-08-04:
When running simulations with Verilator, the following warnings cause errors and halt simulation:

`BLKANDNBLK:` Detected mixed blocking/non-blocking assignments.

`MULTIDRIVEN:` Signal is driven from multiple sources (common in testbenches).

`COMBDLY:` Combinatorial delay path warnings.

These are benign in the context of testbench and simulation flow, but Verilator treats them as fatal unless explicitly suppressed.

Fix:

To resolve this, modify the Makefile to include the following flags:
`VERI_COMPILE_FLAGS += -Wno-BLKANDNBLK -Wno-MULTIDRIVEN -Wno-COMBDLY $(SV_CMP_FLAGS)
`
This ensures Verilator runs without stopping on expected testbench behavior.

Rationale:

Without these suppressions, simulation fails even when the RTL is correct and testbenches are safe. This is blocking users from verifying custom changes (e.g., CVA6 custom instructions, accelerator hooks, etc.).

Please let me know if you would like me to submit a patch or pull request to integrate this fix into the repository. I'm happy to contribute a permanent solution.

Uzair-90, 2025-08-07:
Hi @MikeOpenHWGroup please review the PR #2631 I have provided a fix thank you.

MikeOpenHWGroup, 2025-08-08:
Hi @Uzair-90, #2631 is now merged.  If that resolves this Issue, please close it.