openhwfoundation/core-v-verif #2635: C32E40P/dev: core tests interrupt_bootstrap and interrupt_test don't work with DSIM

djvra, 2025-08-26:
interrupt_bootstrap

e@e:~/core-v-verif/cv32e40p/sim/core$ make dsim-test TEST=interrupt_bootstrap
/home/e/core-v-verif/mk/Common.mk:79: Must define a DPI_DASM_SPIKE_REPO to use the common makefile
/home/e/core-v-verif/mk/Common.mk:353: RISCV set to /opt/riscv
/home/e/core-v-verif/mk/Common.mk:354: RISCV_PREFIX set to riscv32-unknown-elf-
/home/e/core-v-verif/mk/Common.mk:355: RISCV_EXE_PREFIX set to /opt/riscv/bin/riscv32-unknown-elf-
/home/e/core-v-verif/mk/Common.mk:356: RISCV_MARCH set to rv32imc_zicntr_zicsr_zifencei
/home/e/core-v-verif/mk/Common.mk:357: RISCV_CC set to gcc
/home/e/core-v-verif/mk/Common.mk:358: RISCV_CFLAGS set to -march=rv32imc_zicntr_zicsr_zifencei -mabi=ilp32 -O2
mkdir -p /home/e/core-v-verif/cv32e40p/sim/core/dsim_results
mkdir -p /home/e/core-v-verif/cv32e40p/sim/core/dsim_results/dsim_work
*******************************************************************************************
* Cloning CV32E40P RTL model
*******************************************************************************************
git clone https://github.com/openhwgroup/cv32e40p /home/e/core-v-verif/core-v-cores/cv32e40p; cd /home/e/core-v-verif/core-v-cores/cv32e40p; git checkout tags/cv32e40p_v1.8.3
fatal: destination path '/home/e/core-v-verif/core-v-cores/cv32e40p' already exists and is not an empty directory.
HEAD is now at 360d272 Merge pull request #1026 from openhwgroup/dev
*******************************************************************************************
* Compiling with Metrics DSIM
*******************************************************************************************
cd /home/e/core-v-verif/cv32e40p/sim/core/dsim_results && \
        dsim \
        +define+CORE_TB +define+CV32E40P_APU_TRACE +acc+b -timescale 1ns/1ps  \
        -suppress MultiBlockWrite -suppress IneffectiveDynamicCast \
        +incdir+/home/e/AltairDSim/2025/uvm/1.2/src /home/e/AltairDSim/2025/uvm/1.2/src/uvm_pkg.sv \
        -f /home/e/core-v-verif/core-v-cores/cv32e40p/cv32e40p_manifest.flist \
        /home/e/core-v-verif/cv32e40p/tb/core/tb_riscv/include/perturbation_defines.sv \
        /home/e/core-v-verif/cv32e40p/tb/core/tb_top.sv /home/e/core-v-verif/cv32e40p/tb/core/cv32e40p_core_tb_wrapper.sv /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv /home/e/core-v-verif/cv32e40p/tb/core/dp_ram.sv /home/e/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_random_stall.sv /home/e/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_random_interrupt_generator.sv /home/e/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_rvalid_stall.sv /home/e/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_gnt_stall.sv \
        -work /home/e/core-v-verif/cv32e40p/sim/core/dsim_results/dsim_work \
        -genimage dsim.out
Copyright (C) 2017-2025 Altair Engineering Inc. All rights reserved.
=N:[UsageMeter (2025-08-22 18:57:34 +0300)] Using /etc/ssl/certs as location of CA certificates to verify license server.
=N:[UsageMeter (2025-08-22 18:57:35 +0300)] usage server initial connection
=N:[License]
Licensed for Altair DSim Cloud.
=N:[License] New lease granted.
Analyzing...
Elaborating...
  Top-level modules:
    $unit
    cv32e40p_tb_wrapper
    tb_top
    riscv_random_stall
  Found 38 unique specialization(s) of 37 design element(s).
Optimizing...
Building models...
PLI/VPI access: +b
Simulation time precision is 1ps.
  [19/22] module cv32e40p_core_tb_wrapper#(32,22,128,437323776,0,0,0,0,0,1): 368 functions, 4564 basic blocks
  [20/22] module mm_ram#(22,32,32,14,32): 103 functions, 927 basic blocks
Linking dsim.out.so...
/usr/bin/ld: warning: dsim_work/obj/LBuvm_pkg.o: missing .note.GNU-stack section implies executable stack
/usr/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
  Altair DSim version: 2025.0.0000 (b:S #c:32 h:e6a01125e5 os:centos_8)
*******************************************************************************************
* Running with Metrics DSIM
*******************************************************************************************
mkdir -p /home/e/core-v-verif/cv32e40p/sim/core/dsim_results/interrupt_bootstrap && cd /home/e/core-v-verif/cv32e40p/sim/core/dsim_results/interrupt_bootstrap  && \
dsim -l dsim-interrupt_bootstrap.log \
        -image dsim.out \
        -work /home/e/core-v-verif/cv32e40p/sim/core/dsim_results/dsim_work \
         \
        -sv_lib /home/e/AltairDSim/2025/uvm/1.2/src/dpi/libuvm_dpi.so \
        +firmware=../../../../tests/programs/custom/interrupt_bootstrap/interrupt_bootstrap.hex
Copyright (C) 2017-2025 Altair Engineering Inc. All rights reserved.
=N:[UsageMeter (2025-08-22 18:58:05 +0300)] Using /etc/ssl/certs as location of CA certificates to verify license server.
=N:[UsageMeter (2025-08-22 18:58:06 +0300)] usage server initial connection
=N:[License]
Licensed for Altair DSim Cloud.
[... 145 lines cut ...]
=N:Starting event scheduler...
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(265) @ 0ns: reporter [RNDSTALL] INSTR OBI stall enable: 1
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(266) @ 0ns: reporter [RNDSTALL] INSTR OBI stall mode:   1
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(267) @ 0ns: reporter [RNDSTALL] INSTR OBI stall gnt:    0
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(268) @ 0ns: reporter [RNDSTALL] INSTR OBI stall valid:  3
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(269) @ 0ns: reporter [RNDSTALL] INSTR OBI stall max:    0
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(270) @ 0ns: reporter [RNDSTALL] DATA  OBI stall enable: 1
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(271) @ 0ns: reporter [RNDSTALL] DATA  OBI stall mode:   2
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(272) @ 0ns: reporter [RNDSTALL] DATA  OBI stall gnt:    1
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(273) @ 0ns: reporter [RNDSTALL] DATA  OBI stall valid:  0
UVM_INFO /home/e/core-v-verif/cv32e40p/tb/core/mm_ram.sv(274) @ 0ns: reporter [RNDSTALL] DATA  OBI stall max:    1
=E:[UpViolation] unique case violation at /home/e/core-v-verif/core-v-cores/cv32e40p/rtl/cv32e40p_load_store_unit.sv:419:5
  Time: 0ns
  Instance: cv32e40p_tb_wrapper
  No item hit.
TEST 1 - TRIGGER ALL IRQS IN SEQUENCE:
MIP for IRQ[3] not set
tb_top @ 1ns: EXIT FAILURE:               219180
SVA Summary:  1 assertion, 21917 evaluations, 1168 nonvacuous passes, 4 disables
=N:[WriteMetrics] Writing coverage metrics...
=T:Simulation terminated by $finish at time 219180000 (/home/e/core-v-verif/cv32e40p/tb/core/tb_top.sv:143);
Run directory: /home/e/core-v-verif/cv32e40p/sim/core/dsim_results/interrupt_test
  System timescale is 1ps / 1ps
  Altair DSim version: 2025.0.0000 (b:S #c:32 h:e6a01125e5 os:centos_8)
  Random seed: (defaulted to 1)

MikeOpenHWGroup, 2025-08-26:
Hi @djvra, thanks for your interest in OpenHW and the CV32E40P.  As per our documentation, the so called "core" testbench is intended as an example and does not support all tests.  The `interrupt_bootstrap` is an example of such a test.

Having said that, there seems to be a bug with DSim.  When I run that test with DSim 2025.1 on Ubuntu 24.01, I see the same `=E:[UpViolation] unique case violation` that you see.  What version of DSim are you using (and on what platform)?

If you are using DSim, you should be able to use the UVM environment.   Try this:
```
$ cd cv32e40p/sim/uvmt
$ make clean_all
$make test TEST=interrupt_bootstrap SIMULATOR=dsim USE_ISS=0
```

djvra, 2025-08-27:
I got a bunch of the same type of error. Here are some parts of the error:

> Actual argument at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:379:73 replacing formal IN2         spurious '``'
>     From definition of CHECK_PC_EQUAL_LPSTART at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:336:84
>     Expanded from IF_CURRENT_IS_MAIN_HWLOOP at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:379:20
>     Expanded from /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:788:30                           
>     Included from /home/e/core-v-verif/cv32e40p/env/uvme/uvme_cv32e40p_pkg.sv:88:4                                      
> 
>     Actual argument at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:380:73 replacing formal IN2         spurious '``'
>     From definition of CHECK_PC_EQUAL_LPEND at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:337:82
>     Expanded from IF_CURRENT_IS_MAIN_HWLOOP at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:380:22
>     Expanded from /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:788:30                           
>     Included from /home/e/core-v-verif/cv32e40p/env/uvme/uvme_cv32e40p_pkg.sv:88:4                                      
> 
>     Actual argument at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:381:73 replacing formal IN2         spurious '``'
>     From definition of CHECK_PC_EQUAL_LPEND at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:337:82
>     Expanded from IF_CURRENT_IS_MAIN_HWLOOP at /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:381:22
>     Expanded from /home/e/core-v-verif/cv32e40p/env/uvme/cov/uvme_rv32x_hwloop_covg.sv:788:30                           
>     Included from /home/e/core-v-verif/cv32e40p/env/uvme/uvme_cv32e40p_pkg.sv:88:4 

...

>     From definition of RVFI_CSR_IDX_BIND at /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_macros.sv:40:149                unrecognized tokens before ')'
>     Expanded from /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv:334:3                                       
> 
>     /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_macros.sv:41:120                                                        expected 'matches', '&', '&&'... instead of '``'
>     Included from /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_pkg.sv:28:1                                       
> 
>     From definition of RVFI_CSR_IDX_BIND at /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_macros.sv:41:149                unrecognized tokens before ')'
>     Expanded from /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv:334:3                                       
> 
>     /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_macros.sv:42:120                                                        expected 'matches', '&', '&&'... instead of '``'
>     Included from /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_pkg.sv:28:1                                       
> 
>     From definition of RVFI_CSR_IDX_BIND at /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_macros.sv:42:149                unrecognized tokens before ')'
>     Expanded from /home/e/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv:334:3                                       
> 
>   Altair DSim version: 2025.0.0000 (b:S #c:32 h:e6a01125e5 os:centos_8)
> make: *** [/home/e/core-v-verif/mk/uvmt/dsim.mk:140: comp] Error 1